Rustfahrtagentur/microcad #85: cannot call a function from inside a module

fightling, 2024-12-22:
This simple example should work but it doesn't:

[image]

WilstonOreo, 2024-12-22:
[In `ModuleDefinitionBody::eval_statement` in `module_definition_body.rs`.
](https://github.com/Rustfahrtagentur/microcad/blob/f11b641195312c79f4781ca83fc16d1b4e02e6f7/lang/parse/module/module_definition_body.rs#L94)

Somehow it should look like this, by handling the function declaration and adding its symbol to the group.
[image]

fightling, 2024-12-22:
But:

[image]

fightling, 2024-12-22:
@WilstonOreo: Failing tests can be found  [here](https://github.com/Rustfahrtagentur/microcad/blob/pat/improve_module_doc/doc/modules/README.md) and [here](modules/unctions/https://github.com/Rustfahrtagentur/microcad/blob/pat/improve_module_doc/doc/modules/functions.md).

fightling, 2024-12-26:
seems the original example is wrong because of the missing brackets at the module name.

I fixed this, but it's still not working:

[image]

```µcad,test

module m() {
  function f(){}
  f();
}

m();
```

errors: 

```
error: Symbol not found: f
  ---> <no file>:3:3
     |
   3 |   f();
     |   ^
     |
#0	m
	at <no file>:1:1
#1	root
```