Rustfahrtagentur/microcad #96: cannot access module fields from outside as expected

fightling, 2024-12-23:
The following example fails unexpected:

[image]

```txt
error: Symbol not found: i
  ---> <no file>:5:14
     |
   5 | std::print("{i}");
     |              ^
     |
#0	root
error: Cannot nest symbol: Invalid
  ---> <no file>:5:14
     |
   5 | std::print("{i}");
     |              ^
     |
#0	root
```

fightling, 2024-12-23:
Maybe this is related to #45

WilstonOreo, 2024-12-23:
A module without initializers is currently not supported.
`test` should be a namespace.
If `test` is a namespace, the field should be accessed with:
```µcad,namespace_field_test
namespace test {
  i = 1;
}

std::print("i: {test::i}");
```

This example works fine:

```µcad,module_field_test
module test() {
  i = 1;
}

a = test();

std::print("i: {a.i}");
```

fightling, 2024-12-23:
How comes that `i` gets exported as return value instead of a field of `test`?
I find this barely intuitive.