Rustfahrtagentur/microcad #97: Error logs are not reproducable in some cases

fightling, 2024-12-23:
Compiling this example from documentation leads to different output on different runs:

```µcad,hull_single#todo
hull()
    translate(x = [-10, 10]mm, y = [-10, 10]mm)
        circle(1mm);
```

## Output #1

```txt
error: Cannot nest item: translate(CallArgumentList(Refer { value: OrdMap { vec: [CallArgument { name: Some(Identifier(Refer { value: "x", src_ref: SrcRef(Some(SrcRefInner { range: 21..22, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) })), value: ListExpression(ListExpression { list: ExpressionList([UnaryOp { op: "-", rhs: Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 27..29, at: LineCol { line: 2, col: 21 }, source_file_hash: 16566312271860312654 })) })), src_ref: SrcRef(Some(SrcRefInner { range: 26..29, at: LineCol { line: 2, col: 20 }, source_file_hash: 16566312271860312654 })) }, Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 31..33, at: LineCol { line: 2, col: 25 }, source_file_hash: 16566312271860312654 })) }))]), unit: Some(Mm), src_ref: SrcRef(Some(SrcRefInner { range: 25..36, at: LineCol { line: 2, col: 19 }, source_file_hash: 16566312271860312654 })) }), src_ref: SrcRef(Some(SrcRefInner { range: 21..36, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) }, CallArgument { name: Some(Identifier(Refer { value: "y", src_ref: SrcRef(Some(SrcRefInner { range: 38..39, at: LineCol { line: 2, col: 32 }, source_file_hash: 16566312271860312654 })) })), value: ListExpression(ListExpression { list: ExpressionList([UnaryOp { op: "-", rhs: Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 44..46, at: LineCol { line: 2, col: 38 }, source_file_hash: 16566312271860312654 })) })), src_ref: SrcRef(Some(SrcRefInner { range: 43..46, at: LineCol { line: 2, col: 37 }, source_file_hash: 16566312271860312654 })) }, Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 48..50, at: LineCol { line: 2, col: 42 }, source_file_hash: 16566312271860312654 })) }))]), unit: Some(Mm), src_ref: SrcRef(Some(SrcRefInner { range: 42..53, at: LineCol { line: 2, col: 36 }, source_file_hash: 16566312271860312654 })) }), src_ref: SrcRef(Some(SrcRefInner { range: 38..53, at: LineCol { line: 2, col: 32 }, source_file_hash: 16566312271860312654 })) }], map: {Identifier(Refer { value: "y", src_ref: SrcRef(Some(SrcRefInner { range: 38..39, at: LineCol { line: 2, col: 32 }, source_file_hash: 16566312271860312654 })) }): 1, Identifier(Refer { value: "x", src_ref: SrcRef(Some(SrcRefInner { range: 21..22, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) }): 0} }, src_ref: SrcRef(Some(SrcRefInner { range: 21..53, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) }))
  ---> <no file>:1:1
     |
   1 | hull()
     | ^^^^^^
     |
#0	root
```

## Output #2

```txt
error: Cannot nest item: translate(CallArgumentList(Refer { value: OrdMap { vec: [CallArgument { name: Some(Identifier(Refer { value: "x", src_ref: SrcRef(Some(SrcRefInner { range: 21..22, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) })), value: ListExpression(ListExpression { list: ExpressionList([UnaryOp { op: "-", rhs: Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 27..29, at: LineCol { line: 2, col: 21 }, source_file_hash: 16566312271860312654 })) })), src_ref: SrcRef(Some(SrcRefInner { range: 26..29, at: LineCol { line: 2, col: 20 }, source_file_hash: 16566312271860312654 })) }, Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 31..33, at: LineCol { line: 2, col: 25 }, source_file_hash: 16566312271860312654 })) }))]), unit: Some(Mm), src_ref: SrcRef(Some(SrcRefInner { range: 25..36, at: LineCol { line: 2, col: 19 }, source_file_hash: 16566312271860312654 })) }), src_ref: SrcRef(Some(SrcRefInner { range: 21..36, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) }, CallArgument { name: Some(Identifier(Refer { value: "y", src_ref: SrcRef(Some(SrcRefInner { range: 38..39, at: LineCol { line: 2, col: 32 }, source_file_hash: 16566312271860312654 })) })), value: ListExpression(ListExpression { list: ExpressionList([UnaryOp { op: "-", rhs: Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 44..46, at: LineCol { line: 2, col: 38 }, source_file_hash: 16566312271860312654 })) })), src_ref: SrcRef(Some(SrcRefInner { range: 43..46, at: LineCol { line: 2, col: 37 }, source_file_hash: 16566312271860312654 })) }, Literal(Integer(Refer { value: 10, src_ref: SrcRef(Some(SrcRefInner { range: 48..50, at: LineCol { line: 2, col: 42 }, source_file_hash: 16566312271860312654 })) }))]), unit: Some(Mm), src_ref: SrcRef(Some(SrcRefInner { range: 42..53, at: LineCol { line: 2, col: 36 }, source_file_hash: 16566312271860312654 })) }), src_ref: SrcRef(Some(SrcRefInner { range: 38..53, at: LineCol { line: 2, col: 32 }, source_file_hash: 16566312271860312654 })) }], map: {Identifier(Refer { value: "x", src_ref: SrcRef(Some(SrcRefInner { range: 21..22, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) }): 0, Identifier(Refer { value: "y", src_ref: SrcRef(Some(SrcRefInner { range: 38..39, at: LineCol { line: 2, col: 32 }, source_file_hash: 16566312271860312654 })) }): 1} }, src_ref: SrcRef(Some(SrcRefInner { range: 21..53, at: LineCol { line: 2, col: 15 }, source_file_hash: 16566312271860312654 })) }))
  ---> <no file>:1:1
     |
   1 | hull()
     | ^^^^^^
     |
#0	root
```

Seems that the nodes come in different orders.

WilstonOreo, 2024-12-23:
The `hull()` algorithm is not implemented.
It might also be that we use a BTreeMap instead of a HashMap for the SymbolTable. I noticed that HashMap is not deterministic due to the underlying hashing algorithm. This is good for security but bad for reproducibility.

fightling, 2024-12-23:
We can just sort this when displaying.

fightling, 2024-12-23:
seems solved

fightling, 2024-12-23:
seems solved

closed 2024-12-23 by fightling