Rustfahrtagentur/microcad #99: structure of `std` and `builtin`

fightling, 2024-12-23:
I just moved some builtin stuff in `std` into a sub folder `builtin`, then removed some obsolete files - and look what happened.
The module is empty beside the `std.µcad` file which one would expect in the `std` folder - somehow.

Previous structure was:

- `std`
  - `std.µcad`
  - builtin stuff
  - obsolete files
  - `Cargo.toml` etc.

now it's like 

- `std`
  - `std.µcad`
  - `builtin`
    - builtin stuff
  - `Cargo.toml` etc.

which looks awful!

I would suggest to have two crates: `builtin` and `std`.

`builtin` shall include the stuff I moved and the `Cargo.toml`, and `std` just the µcad files.
Because one want to avoid writing `std::std` somewhere, it may be useful to have everything in separate files, like:

- `std`
  - `algorithm.µcad
  - `math.µcad
  - etc.

...and because we might want to have multiple libraries and also not want to scan our whole project directory for µcad sources, we might want to have a lib directory.

- `lib`
  - `std`
    - `algorithm.µcad
    - `math.µcad
    - etc.
  - additional libs
  
Unclear to me how the git submodule structure must look like.

fightling, 2024-12-24:
saw that this is a doublet of [64](https://github.com/Rustfahrtagentur/microcad/issues/60. closing the older one.

fightling, 2024-12-24:
only thing left to talk about is how to bring the std-library into our repository.

Because at some point we will have distro packages where including additional folders may no problem.

The standard library should be stored in a separate repository.

Don't really know how to bring this all together.

WilstonOreo, 2024-12-24:
Suggestion: How about we put the std into a separate repo and version it. The cli version specifies a std lib version. By default, cli and std versions should match (e.g. 1.0.0). When the CLI 1.0.0 starts, it looks in a configurable library directory (e.g. ~/.microcad) whether the std 1.0.0 is installed. If it is not installed, it is downloaded from the repo and put into the lib folder. 
This way, we only need to ship the cli executable and the std and any other libs are installed when required.

WilstonOreo, 2024-12-24:
Another suggestion: What about integrating the entire std into the microcad-cli binary via `include_str!()`?
Also, I like to have module-like structure as in Rust. Assume we have a folder called `std`. If the folder contains a file `lib.µcad`, it can be imported as library with `use std;`.
The folder structure looks like this:

- `std`
  - `lib.µcad`
  - `geo2d.µcad`
  - `geo3d.µcad`
  - `algorithm.µcad`
  - ...

`lib.µcad` include the other files e.g. via `pub use geo2d`, and so on.

closed 2024-12-24 by WilstonOreo

fightling, 2024-12-25:
I've closed this issue but wanna continue this discussion in #105

PR #100 (2024-12-25): std and builtin
closes #99