Rustfahrtagentur/microcad #102: Tuples

fightling, 2024-12-24:
I wanna start a discussion about our superhero-kind of tuples.

They can:

- include unnamed items
- have an order
- include named items
- items can be accessed by name
- they can be matched to the parameters of a module or function 
  - e.g. both `(x, y, size(w, y))` and `((x, y), (w, y))`  match `f(x, y, w, h)` where `size` would dissolve.
- items are *not* accessible by position (this is in arrays)

Only the first two behaviors are related to what programmers call a *tuple*.

So at first I would like to find another name which is more about what this data structure really is:
A set of parameters.
So I suggest to name it `Parameters` (or something similar).

In the documentation I need to describe the versatile character of `tuples` and I a stuck at getting it into a sectional description.

That's why I start this discussion.

WilstonOreo, 2024-12-26:
Not sure about renaming the tuples to *Parameters*, but I agree to the rest.
Here is a more specific use case:

Assume you have a `module cube(x: Length, y: Length, z: Length)`.
You can call the module in the following ways:

```µcad
cube(1cm,1cm,1cm);
v = (x = 1, y = 1, z = 1)cm;
cube(v);
```

Because `v` is a vec3 with fields `x`, `y`, `z` and the implicit initializer of `cube` has the same fields with the same types, you can pass `v` as an argument to `cube`.  `x`, `y`, `z` will be substituted with the fields of `v`.

fightling, 2024-12-26:
Wikipedia:

> In [computer science](https://en.wikipedia.org/wiki/Computer_science), tuples come in many forms. Most typed [functional programming](https://en.wikipedia.org/wiki/Functional_programming) languages implement tuples directly as [product types](https://en.wikipedia.org/wiki/Product_type),[[1]](https://en.wikipedia.org/wiki/Tuple#cite_note-2) tightly associated with [algebraic data types](https://en.wikipedia.org/wiki/Algebraic_data_type), [pattern matching](https://en.wikipedia.org/wiki/Pattern_matching), and [destructuring assignment](https://en.wikipedia.org/wiki/Assignment_(computer_science)#Parallel_assignment).[[2]](https://en.wikipedia.org/wiki/Tuple#cite_note-3) Many programming languages offer an alternative to tuples, known as [record types](https://en.wikipedia.org/wiki/Record_(computer_science)), featuring unordered elements accessed by label.[[3]](https://en.wikipedia.org/wiki/Tuple#cite_note-4) A few programming languages combine ordered tuple product types and unordered record types into a single construct, as in [C structs](https://en.wikipedia.org/wiki/Struct_(C_programming_language)) and Haskell records. [Relational databases](https://en.wikipedia.org/wiki/Relational_database) may formally identify their [rows](https://en.wikipedia.org/wiki/Row_(database)) (records) as tuples.