Rustfahrtagentur/microcad #116: Anonymous modules

fightling, 2025-04-15:
### Hint 

Relates to work in branch `/pat/project`:

### Problem

Sometimes one might be forced to have multiple modules within the same file to get a specific namespace structure.
This might lead to very long files and there seems no escape from that.

For example if you want to have a module called `std::geo2d::circle` you cannot get this by using `std/geo2d/circle.µcad`.
Instead you would end up with a name `std::geo2d::circle::circle` because you cannot construct a module at source file level without declaring a name.

### Solution

The Solution might be to have anonymous modules which could be declared like in the following file `std/geo2d/circle.µcad`:

```µcad
module(radius=1) { }
```

Which SHALL lead to the correct name `std::geo2d::circle` then.

### Consequences

On might be motivated to write:

fightling, 2025-04-15:
Other idea: Use `mod.µcad` like in rust.

This would solve the problem.

WilstonOreo, 2025-04-21:
I think we can close this?

closed 2025-05-16 by fightling