Rustfahrtagentur/microcad #121: Built-in Methods for values and objects

WilstonOreo, 2025-04-22:
Some built-in methods for values might be useful.
`MethodCall::eval` needs to be implemented to achieve this.

### Values

- `count()`: Number of elements list or a string
- `abs()`: Calculate absolute value. Calls `__builtin::abs` internally.

### Object Nodes

A generated object nodes should have built-in methods to request certain geometry information.
E.g., I would like to retrieve the volume from a sphere (`sphere(r = 3.0).volume()`).

I can currently think of the following built-in methods:
- `area()`: Calculate the surface area of a geometry node.
  To calculate the area, a 2D geometry needs to be rendered into a MultiPolygon and a 3D geometry into a Mesh, respectively. 
- `volume()`: Calculate the volume of a 3d geometry node
  To calculate the volume, a 3D geometry into a Mesh, respectively. 
  Returns 0 for 2D geometries.
- `points()`: Return a list of points of geometry. E.g. for a rectangle, it returns its 4 corner points. 
```
// Get 6 points from a hexagon
points = ngon(radius = 10.0, n = 6).points();
assert(points.count() == 5);
// Construct 6 circles at the hexagon's points via multiplicity
circle(radius = 3.0, center = points);
```
- `edges()`: A list of point pairs containing all edges of a geometry 

I would like to have minimum list of methods for now but with the possibility to extend the built-in-methods in the future.

WilstonOreo, 2025-06-02:
`export` should be an automatic builtin method for every module:
```
std::geo3d::cube(40mm).export("cube.stl","cube.png");
```

But if we want to have additional parameters (e.g. resolution for png files) we cannot have this syntax but maybe:
```
std::geo3d::cube(40mm).export("cube.stl").export("cube.png",resolution=100dpi);
```

This would mean that export must pass-thru the object nodes to the second call of export.

fightling, 2025-06-04:
related to #136