Rustfahrtagentur/microcad #122: Object attributes for export and visualization

WilstonOreo, 2025-04-22:
In the Object tree resulting from an `eval`, it is useful to assign attributes to each object, e.g. in order to tag them with name, assign a color for the visualizer or mark them for export.
Attributes only affect the object nodes.

### Attribute syntax

We need to have a syntax for attributes. I suggest using a subset of [Rust outer attributes](https://doc.rust-lang.org/reference/attributes.html), e.g.: `#[hide], #[export("file.svg")], #[color(std::colors::red)], #[svg::style(stroke_color = #FF00FF, fill_color = std::colors::blue)]`

The grammar rule might look as follows:
```
attribute = { "#[" ~ ws* ~ (call | qualified_name)  ~ ws* ~ "]" }
```

Only the following the statements can have attributes:

- Assignment with attribute:
```
#[color(red)]
cube = std::geo3d::cube(a = 3.0); // The cube will be red
```

- Expression with attribute:
```
// Hide this sphere in the visualizer
#[hide]
sphere(r = 4.0);
```

Attributes might be wrapped into modules:
```
module colorize(color: Color) {
    #[color(color)] {
        @children
    }
}
```

### Extending `objects::Object`

We need to extend the `Object` struct:

```
struct Attributes(HashMap<Id, CallArgumentValueList>);

struct Object {
    /// Attributes
    pub attributes: Attributes,
    ...
}
```

### Export attributes

The following code will produce an object node containing a sphere geometry.
The object node's attributes will have an entry "export" with a nameless argument "sphere.stl" as call argument value list.

```
#[export("sphere.stl")]
sphere(r = 3.0);
```

When this source file is processed by the CLI using the `export` command, it will scan all top level object nodes for the `export` attribute and export them to the respective file.

fightling, 2025-04-22:
Why not follow the material path (see https://github.com/Rustfahrtagentur/microcad/issues/41) instead of color or visibility attributes. And why shall I hide Also can this attribute be overridden by top level attributes?

I see that you need the `export` thing but I would just stay with a `__builtin::export()`

WilstonOreo, 2025-04-22:
I see some shortcomings in the approach #41:
 * We have to introduce a special type `Material`, unclear for now how it will look like 
 * another syntax rule for `use`
 * a new keyword `material` and new Statement enum variant
 * Unclear order of execution: Will a material statement affect all following statements? Will the material statement only affect the top level or all sub-nodes?
 * asking the objects for attributes via method calls, so there is no clear distinction between properties, method and materials in the end.
 * We should split geometry and material, it least for primitives in the std lib. You often want to render the same geometry with a different material. In the visualizer, you can render the same geometry and apply different material shader for each instance. This difficult to do when material and geometry are not separated.

I currently see the approach above more flexible and easier to implement.
Materials can be added later to the attribute system and wrapped in a module `material`:

```
module material(m: Material) {
     #[material(material)] {
          @children
     }
}

// A PLA circle
material(m = polymers::PLA(color=#00FF00, density=1kg/m³)) circle(r = 3.0);
```

Regarding the `export`, we currently have the `Export(ExportSettings)` variant in `ObjectNodeInner`.
Instead of scanning for Export nodes, we search for objects containing an `export` attribute. The export variant can be removed. Instead, we can define a module `export` in the std lib via:
```
module export(filename: String) {
   #[export(filename)] {
      @children
   }
}

std::export("test.stl") sphere(r = 3.0);
// This would enable multiplicity as well:
std::export(["test.stl", "test.ply"]) sphere(r = 3.0);
```

fightling, 2025-06-03:
I still see attributes like `color` or `hide` as state of the viewer which surely should be permanent when needed (like saving a view or restoring after closing the editor).

The approach with attributes in code is something which openscad is doing and I would prefer not to have view attributes in code.

But maybe both could also be useful.

The material approach (I have already made some suggestions) could be implemented and bring some material related tooling into µcad code. 

I would like to introduce a new keyword `material` which is kind of a module but which does not produce any object nodes and instead interacts  with them by overwriting an object node's material properties.

Material properties may be quite advanced (see https://3d.nice-cdn.com/upload/file/Bambu_ABS_Technical_Data_Sheet_V3.pdf).

Here is another sketch of what I am currently imagining:

```µcad
namespace std {
    material Material(density: Density) {
        function weight() {
           density * object.volume();   // implicit access to object (aka module) functions and properties
        }
    }
  }

namespace Bambu {

   material Bambu( max_overhang_angle: Angle, max_bridging_length: Length)  : std::Material {   //  inherits density  from std::Material

      function check_overhang(angle: Angle) {
         return angle < max_overhang_angle;
      }
   }
 
   namespace ABS {
    
      // base ABS material
      material ABS(color: Color) : Bambu::Bambu {   // inherits check_overhang() from Bambu::Bambu
        density = 1.05 g/cm³;
        max_overhang_angle = 70°;
        max_bridging_length = 40 mm;
     }
      
    material Black : ABS { 
        name = "ABS Black, Refill (1.000 g)";    // set vendor's name
        color = #000;    // overwrites color of ABS
     }

  }

}

box = std::geo3d::cube(10mm).material(Bambu::ABS::Black);   // overwrite material of object node

std::print("{box.name}: {box.weight()}");
```

Materials may be overwritten by superior modules or by the user in the UI.

I see there is complexity but on the other side `material` is quite like a module but can be derived and inherits properties which is the most challenging thing about it.

But we can stay with attributes until this gets implemented. 

But I see advantages for our presentation at PTF when one can label materials of sub objects.

WilstonOreo, 2025-06-03:
Can we close this for now?

closed 2025-08-05 by fightling